NyanCAD/Mosaic #12: Basic subcircuit support

ArthurAdmiraal, 2021-08-30:
- [ ] Must have variable number of terminals
- [ ] Should have variable number of terminals per side
- [ ] Must have unique terminal names
- [ ] Should have user-settable terminal names
- [ ] Must have use-settable model name
- [ ] Should display display model name

pepijndevos, 2021-08-30:
Tricky thing is, how do you map the port names to the subcircuit port order. Here too a model needs to be able to specify its spice representation. Exposing this as a textfield is horrible UX but might work, but need a better system.

pepijndevos, 2021-09-11:
Key thing I need to refactor: allow setting and changing the loaded schematic.

pepijndevos, 2021-09-13:
Schematic can now be dynamically changed.

pepijndevos, 2021-09-14:
Big question is, how do I want to structure subcircuits?

On the one hand I want the flexibility of Cadence/VHDL-like entity -> implementation abstraction, where different schematics can implement the same entity, and be swapped out when configuring a simulation.

On the other hand the common case is probably just one implementation per entity, and I want to have the simple UX of just being able to "drill down" in a schematic, and run a simulation without having to specify every single component.

Maybe it'd work to just build the UI for the single-implementation case, and leave it up to the simulator to do substitutions. So like, you can just simulate the schematic as drawn, or you can provide a configuration which will override subcircuits and parameters.

pepijndevos, 2021-09-14:
Pffffew... It's starting to come together. You can now specify the background and port layout, and select an image URL on top of that.

There is a bug that causes the image to be dragged https://stackoverflow.com/questions/69178308/how-to-prevent-dragging-of-an-image-element-inside-a-svg

And spice generation needs a complete overhaul. To begin I need to refactor the code that gets the element ports and matches them to wires, potentially using the index I described above. Then I actually need to add code to print the `x` elements and the `.subckt` elements.

pepijndevos, 2021-09-14:
Ok, bug fixed, edit links added for peak drill-down performance.

Spice export still bust. Broken version up at http://wishfulcoding.nl/mosaic

closed 2021-09-21 by pepijndevos