NyanCAD/Mosaic #57: hierarchy: categories and layouts

pepijndevos, 2022-04-10:


pepijndevos, 2022-04-11:
Also consider the cell/model selection behavior once models have a "path" #78

jellebakker96, 2022-04-19:
Also consider that it would be nice if you could reference the same cell in multiple categories and that it might be desirable to add support for sub, subsub, subsubsub, etc catogories later on.

pepijndevos, 2022-04-19:
Hmmm why would you want the same cell in multiple categories? I mean, it's not hard to do, but could be confusing. Like, how would you have a good UX for "copying" a cell to another path, and make a distinction between a copy and an alias.

I guess in windows the equivalent is "create shortcut" and in Linux you have symbolic links and hard links. Where symbolic links are a little file that points to the original, while a hard link is just... two paths that refer to the exact same data.

It's a nice feature, I'm just concerned how to make it not confusing.

jellebakker96, 2022-04-20:
I just looked into it a bit more but it looks like I was mistaken. The main cell you copy does not become a shortcut. I was thinking of the cells used inside the copied cell that still reference their original cell instead of also being copied to the new location (this is called hierarchical copy).

Sorry for the confusion.

jellebakker96, 2022-04-20:
You might want to use the same cell in multiple designs. An example could be an opamp that is used in multiple locations which you then might want to add to for example the "buffer_1" and "buffer_2" categories. 

This ability to add cells to multiple categories is also why I think cadence gives each library the default categories "Everything" and "Uncategorized".

jellebakker96, 2022-04-20:
This also means that you can reference any cell if you know its name and its library. You do not have to know the category.

jellebakker96, 2022-04-20:
In cadence, I do not find this confusing because the name of each cell in each library is unique so it is quite clear that the category is only a way to order your cells. The category does not have any other feature.

commit f956ffc6 (2022-05-25): hierachy, fixes #57
closed 2022-05-25 by pepijndevos