NyanCAD/Mosaic #75: See video for strange drag behavior

jellebakker96, 2022-04-10:
https://user-images.githubusercontent.com/35449392/162637067-4e522fc6-6e70-4aff-b2e8-0da7897bae77.mp4

pepijndevos, 2022-04-11:
Same bug as  #63

pepijndevos, 2022-04-11:
Hmmm can't actually reproduce this problem on either Firefox or Chrome. Can you still reproduce this with the fixes in #63 and #58 in place?

pepijndevos, 2022-04-11:
And, do you need to do anything special? Like, does it only happen when zoomed out or after some other action, or is dragging always at double the speed?

jellebakker96, 2022-04-13:
No, I do not need to do anything special. It happens when I am zoomed out and when I am not zoomed out. It also happens on both my laptop and PC. My pc has a 4k display and my laptop has 1080p plus 5120x1440p displays.

jellebakker96, 2022-04-13:
I also see that sometimes when I am dragging that suddenly the zoom-in function gets enabled.

jellebakker96, 2022-04-13:
This zoom-in function then keeps zooming in as long as I keep the middle mouse button pressed

pepijndevos, 2022-04-13:
I've tried a random fix in 3fb4d56facf7639f7a45ca940e58f10640e7cc96 does that help?

pepijndevos, 2022-04-13:
:(

Also works fine on Sonya's laptop. straaaange.

pepijndevos, 2022-04-13:
I figured it out!!! Do you have your display setting set to more than 100%? I see the same behavior when I set my scaling to 200%

jellebakker96, 2022-04-13:
Yes

jellebakker96, 2022-04-13:
All, my displays have the scaling set to >100%

jellebakker96, 2022-04-14:
I now know what causes the zoom-in bug. First start dragging with your middle mouse button and then (while still having pressed your middle mouse button) start scrolling to zoom out. Instead of zooming out, you will start to zoom in.