NyanCAD/Mosaic #77: Make colour order consistent

jellebakker96, 2022-04-11:
I would have thought that the color of the last added component should be on top when it is placed right next to another component but this seems to not be the case. I do not know how the color order is determined right now, but to me it seems inconsistent.

pepijndevos, 2022-04-11:
???????

jellebakker96, 2022-04-11:
[image]

I would have thought that the order in which you place components changes which color is on top. This does not seem to be the case. How did you determine which color is on top?

jellebakker96, 2022-04-11:
If you place three capacitors with one cell empty in between them and then place nmos in those empty cells some of the blue is above and some of it is below the green from the capacitors

pepijndevos, 2022-04-11:
Whatever order they are drawn in, which is whatever order they are in the map, which is arbitrary, so order is arbitrary and this is not going to change any time soon.

closed 2022-04-11 by pepijndevos

jellebakker96, 2022-04-11:
Ok