NyanCAD/Mosaic #78: Component location can change thereby breaking the "add nmos/pmos/pnp/npn/etc" model selection drop down menu

jellebakker96, 2022-04-11:
When you have a second library you should be able to say for each component if it can be added by "add subcircuit" or "add PNP BJT/NPN BJT/NMOS/PMOS/RES/etc" such that the new model becomes available in the drop-down menu of the correct component. Otherwise, that entire sidebar becomes useless. 

This functionality will also be needed for when you have for example pmos/nmos devices without bulk port that have their own sub symbol in the future. 

What would happen for example to the "PNP BJT" model drop-down list if the desired PNP BJT is not in the default "pmos" category/cell?

Richt, now it seems like you simply show all models that are inside a couple of the default catogories/cells like (diode/pmos/nmos/etc), but that seems really limited to me.

jellebakker96, 2022-04-11:
Maybe also add a drop-down field for library selection. Such that you could search for all pmos devices in a specific library. If the library field is blank it would simply show all pmos devices in the library manager.

pepijndevos, 2022-04-11:
I have no idea what you mean. Are you talking about #67 ?

pepijndevos, 2022-04-11:
So it's about #57 ? Like, you want /mylib/mypdk/coolmos too only show other coolmos in the dropdown?

jellebakker96, 2022-04-11:
Yes, it is related to that.

jellebakker96, 2022-04-11:
There should be some way to set which models are shown in the dropdown model.

pepijndevos, 2022-04-11:
Ok, I'll close this one and note it under #57

closed 2022-04-11 by pepijndevos