NyanCAD/Mosaic #84: Simulation progress bar

jellebakker96, 2022-05-07:
Maybe also show simulations before they are finished?

pepijndevos, 2022-05-07:
It already streams data from the simulator, so the only part where you don't see progress is while it's loading the schematic into the simulator.

pepijndevos, 2022-05-07:
I think error reporting can be improved in cases where there is a problem with connecting to the simulator, but since you already get streaming data as soon as the simulation has started, and can watch the simulator output in the terminal window at the bottom, I don't see any value in a progress meter.

closed 2022-05-07 by pepijndevos

jellebakker96, 2022-05-08:
Ah, ok

jellebakker96, 2022-05-08:
> I think error reporting can be improved in cases where there is a problem with connecting to the simulator, but since you already get streaming data as soon as the simulation has started, and can [...]

This was indeed the problem.