NyanCAD/Mosaic #104: gdsfactory integration

joamatab, 2023-01-24:
How could we integrate Mosaic with [gdsfactory](https://github.com/gdsfactory/gdsfactory)

Mosaic could be a great for spice simulations and schematic capture

https://github.com/gdsfactory/gdsfactory/issues/681  spice simulator
https://github.com/gdsfactory/gdsfactory/issues/934 schematic capture

pepijndevos, 2023-01-25:
What do you imagine this integration to look like?
What could Mosaic offer that makes it better suited for use with gdsfactory?

Mosaic stores schematics in a JSON format in CouchDB and is able to generate spice from there.
It would for example be possible to generate code or Yaml from the same data.

I'm particularly interested in how it could work well with schematic driven layout.

pepijndevos, 2026-04-16:
wip